TOLDOTECHNIK/buildroot-webkit #5: url.txt not loading

stoehrmark, 2019-01-19:
I have used your prebuilt image RPi 3 B (2018-12-01) on a Pi3B+, I added the WIFI config and also added a file /boot/url.txt 777. But cannot get it to load the webpage? In the url.txt I added the URL (http://www.mysiteexample.com/index.php) on the first line but the Pi always boots to the orange screen with the moving squares every time? I can access the web inspector so i know the Pi is online.

Any help would be appreciated.
Thanks
And great work on this software!!

TOLDOTECHNIK, 2019-01-20:
It seems, you're still using the original url.txt Try connecting to your RPi using SSH. Then do the following:
`nano /boot/url.txt`
On the first line `http://localhost` should be listed. Replace this with your own url. Then save the file by typing `ctrl-x` and `y` After that do a reboot.

stoehrmark, 2019-01-20:
Magic! Thank you.
I was editing the file with the sdcard in a card reader connected to a Ubuntu machine, not via serial SSH directly on the Pi

Now works great. Plays video nice and smooth

New question
How often will it check for a change in the URL? If I change the page will it update the player?
Or can I force an update on the Pi remotely

Thanks

stoehrmark, 2019-01-20:
Can it be may be made to automatically refresh the URL on a timing? Maybe every 5 - 10  min for example.

TOLDOTECHNIK, 2019-01-21:
> I was editing the file with the sdcard in a card reader connected to a Ubuntu machine

This should work. Maybe you removed the card before the changes were written onto it.

> How often will it check for a change in the URL?

The URL gets loaded once every reboot.

> Can it be may be made to automatically refresh the URL on a timing? Maybe every 5 - 10 min for example.

This feature is not implemented.

closed 2019-01-21 by TOLDOTECHNIK