TOLDOTECHNIK/buildroot-webkit #8: Enable touch

roderick86, 2019-04-02:
Hi guys,
Great work!
Just wondering if you have a solution for getting touch to work with the Raspberry Pi Zero W image. The display/touchscreen is an Adafruit ADA-2260. Already tried to uncomment `export WPE_BCMRPI_TOUCH=1` in /usr/bin/WPE but no results so far. Any assistance would be very kind.
Thanks in advance.

TOLDOTECHNIK, 2019-04-02:
Thanks.
Adafruit's ADA-2260 touch screen acts as a USB mouse. We think you have to enable the `WPE_BCMRPI_CURSOR` setting. Please see: https://github.com/TOLDOTECHNIK/buildroot-webkit#keyboard-and-mouse-input

roderick86, 2019-04-03:
Hi, unfortunately it didn´t work.

Just installed raspbian on a second microSD and touch is working fine, any
suggestions?

Thanks once again.

TOLDOTECHNIK, 2019-04-03:
If we got it right your display is using the AR1100 for the USB HID device. Could you try to run `dmesg | grep AR1100` on your RPi and check if there's some output?
If not, we have to see which exact driver is needed for it.

roderick86, 2019-04-03:
# dmesg | grep AR1100
[    2.460838] usb 1-1: Product: AR1100 HID-MOUSE
[    2.481098] input: Microchip Technology Inc. AR1100 HID-MOUSE as /devices/platform/soc/20980000.usb/usb1/1-1/1-1:1.0/0003:04D8:0C02.0001/input/input0
[    2.495987] hid-generic 0003:04D8:0C02.0001: input,hidraw0: USB HID v1.11 Mouse [Microchip Technology Inc. AR1100 HID-MOUSE] on usb-20980000.usb-1/input0

So I guess driver is found...

TOLDOTECHNIK, 2019-04-03:
No good news. Do you see the mouse cursor?

roderick86, 2019-04-03:
Mouse cursor is visible and works with a normal mouse..

TOLDOTECHNIK, 2019-04-03:
We're sorry. Unfortunately we don't have your hardware here to investigate the problem. Maybe someone else knows more?

TOLDOTECHNIK, 2019-04-26:
Closing due to no activity.

closed 2019-04-26 by TOLDOTECHNIK