TOLDOTECHNIK/buildroot-webkit #9: missing python

gaby64, 2019-04-24:
I would need python to install onoff node module to control gpio

there seems to be no way to install or build from source anything from within the pre-built images.

TOLDOTECHNIK, 2019-04-24:
That's correct. The slimmed down distribution is not meant to compile anything on it. But there's a way to implement the `onoff` Node.js module.

If you want to build the image on your own, you have to do the following:
- remove the "custom" Node.js binary (`rm board/toldotechnik_rpi/rootfs-overlay/usr/bin/node`), so the buildroot make process will compile the `onoff` module against the correct Node.js version
- Enable `python` (`make menuconfig` ->  Target packages -> Interpreter languages and scripting -> python)
- Add `onoff` module (Target packages -> Interpreter languages and scripting -> nodejs -> Additional modules -> add `onoff`)
- Recompile everything with: `make clean && make`

If you're lazy ;-), you can download a precompiled RPi 3 image here: https://dev.toldotechnik.li/download/387828/

Be sure to `export NODE_PATH=/usr/lib/node_modules` before testing your Node.js application in the terminal.

gaby64, 2019-04-24:
thanks, im definitely taking the lazy route, I dont have a linux machine

TOLDOTECHNIK, 2019-04-26:
Did you have success with the new build? So we can close the issue.

gaby64, 2019-04-26:
yes, altho 1080p playback is choppy and playing such video file from the microsd card leads to complete system unresponsiveness.

Raspberry Pi 3 B+

I decided to just run it as a server and display the webpage on a laptop

the intent is to show a quarry blast and activate a relay to shake the platform on which people stand.

closed 2019-04-26 by TOLDOTECHNIK