TOLDOTECHNIK/buildroot-webkit #13: How start apache2 server?

MaximMikulich, 2019-09-04:
I install apache2 and php, but window white.....

TOLDOTECHNIK, 2019-09-04:
Try to display the PHP info message in your index file:
```
<?php
phpinfo();
?>
```
If there's no output, have a look in the /etc/init.d folder. Is there a init script for apache/php? Is the apache server running? Check the output of `htop` command on the fly.
We cannot get you more support on this.

MaximMikulich, 2019-09-04:
In the /etc/init.d folder haven't scripts for apache/php

TOLDOTECHNIK, 2019-09-04:
Try to use this script: https://github.com/buildroot/buildroot/blob/master/package/apache/S50apache
Put it in the `/etc/init.d` folder and make it executable `chmod +x S50apache`
Maybe you need something similar for PHP. We're not familiar with it.

MaximMikulich, 2019-09-04:
@TOLDOTECHNIK ```
httpd not running, trying to start
(98)Address already in use: AH00072: make_sock: could not bind to address 0.0.0.0:9998
no listening sockets available, shutting down
```

TOLDOTECHNIK, 2019-09-05:
Try removing `S30nodeserver` and reboot again.

MaximMikulich, 2019-09-05:
in /var/logs/error_log
```
[Thu Sep 05 11:19:22.927420 2019] [mpm_event:crit] [pid 0:tid 1996308464] (70023)This function has not been implemented on this platform: AH00495: Couldn't create a Thread Safe Pollset. Is it supported on your platform?Also check system or user limits!
[Thu Sep 05 11:19:22.927866 2019] [:emerg] [pid 0:tid 1996308464] AH00017: Pre-configuration failed, exiting
```
httpd not running

TOLDOTECHNIK, 2019-09-05:
Okay, we can't provide any more support from here.

TOLDOTECHNIK, 2019-09-09:
### Apache2 ###
Apache2 with PHP works absolutely fine. For the setup there are a few things you have to keep in mind. To make it simple, here's a short step to step guide:

`make menuconfig`
Target packages -> Networking applications -> [x] apache (default settings)
Target packages -> Interpreter languages and scripting ->
  [x] php
  │ │ [x] FPM interface
`make`

Write the image to the SD card. Then do some adjustments on your Raspberry Pi
`rm /etc/init.d/S30nodeserver`
`nano /etc/init.d/S50apache`
```
#!/bin/sh
#
# Start the apache service ...
#

rm /var/logs/httpd.pid   #Quick and dirty fix for: Apache error “Error retrieving pid file /var/logs/httpd.pid”

APACHE_DIR=/usr/bin

case "$1" in
  start)
    echo "Starting apache web server..."
    $APACHE_DIR/apachectl -k start
    ;;
  stop)
    echo -n "Stopping apache web server..."
    $APACHE_DIR/apachectl -k stop
    ;;
  restart|reload)
    $APACHE_DIR/apachectl -k stop
    $APACHE_DIR/apachectl -k start
    ;;
  *)
    echo $"Usage: $0 {start|stop|restart}"
    exit 1
esac

exit $?
```
make the script executable with `chmod +x /etc/init.d/S50apache`

To enable PHP you need to modify the httpd.conf file

`nano /etc/apache2/httpd.conf`

Add following to the `<Directory "/usr/htdocs">` node
    `DirectoryIndex index.php`
and insert this at the end of the file:
```
<FilesMatch ".+\.ph(p[3457]?|t|tml)$">
  SetHandler application/x-httpd-php
</FilesMatch>
```

This is the folder for the web files: `/usr/htdocs/`

To check if PHP works a test file can be created.
`nano /usr/htdocs/index.php`
```
<?php
phpinfo();
?>
```

Restart the Apache web server by calling `/etc/init.d/S50apache restart` or reboot the device.

You can put all the modified files into the board/toldotechnik_rpi/rootfs-overlay/ directory so that they are included after every build of the image.

### NGINX ###
However, we recommend using the NGINX web server. It is much faster and comes already with a predefined init script (`/etc/init.d/S50nginx`).

`make menuconfig`
Target packages -> Networking applications -> [x] nginx (default settings)
Target packages -> Interpreter languages and scripting ->
  [x] php
  │ │ [x] FPM interface

`make`

Write the image on the SD card.

This is the folder for the web files: `/usr/html`

To enable PHP you need to modify `/etc/nginx/nginx.conf`

`nano /etc/nginx/nginx.conf`

Put this in the http `{ server { ... } }` node.
```
 location ~ \.php$ {
     fastcgi_pass  unix:/var/run/php-fpm.sock;
     include       fastcgi.conf;
 }
```
If `index.php` should be called by default, adjust the location node as follows.
```
location / {
  root   html;
  index  index.php index.html index.htm;
}
```

`/etc/init.d/S50nginx restart`

(___In return we would be happy if you would show us a short video of your project you currently working on.___)

MaximMikulich, 2019-09-09:
@TOLDOTECHNIK I did) rebuilt and it worked) But thanks for the detailed description! It will help someone!

closed 2019-09-10 by TOLDOTECHNIK

Rdivxe, 2021-12-07:
@TOLDOTECHNIK i want NGINX/PHP image for "RPi0 w" pleas