TOLDOTECHNIK/buildroot-webkit #14: Unable to ssh in prebuilt PiZero image

retrorich75, 2019-09-28:
use command prompt ssh or Putty 

Unable to connect using ssh in prebuilt image RPi Zero / Zero W (2019-06-24)

Mouse and Keyboard combo not working unable to check if  Mouse support in the browser window is enabled by uncommenting the export WPE_BCMRPI_CURSOR=1 line in /usr/bin/wpe

since no ssh connection

TOLDOTECHNIK, 2019-09-28:
Did you properly edit the wpa_supplicant.conf file on the /boot partition so that your Zero can connect to the wireless network?
https://github.com/TOLDOTECHNIK/buildroot-webkit#wifi-configuration

retrorich75, 2019-09-28:
> Did you properly edit the `wpa_supplicant.conf` file on the `/boot` partition so that your Zero can connect to the wireless network?
> https://github.com/TOLDOTECHNIK/buildroot-webkit#wifi-configuration

Yep, could load a webpage too with url.txt

TOLDOTECHNIK, 2019-09-28:
So, you probably need to get your RPi's IP address to access it's SSH.

bartbergman, 2019-10-06:
you probably need to place an empty file called `ssh` in the `boot` folder (next to `cmdline.txt`, `config.txt` etc.)

TOLDOTECHNIK, 2019-10-07:
Putting the empty `ssh` file into the `/boot` folder is related to the current Raspbian distro. Our ssh server is enabled by default, no additonal configuration is needed.

TOLDOTECHNIK, 2019-11-07:
Closing due to no activity.

closed 2019-11-07 by TOLDOTECHNIK

hortopan, 2020-01-18:
I can confirm this is still happening.  I can't SSH into the prebuilt image although it properly boots and the RPI 0 w connects to wifi. Doesn't look like 22 is opened.

TOLDOTECHNIK, 2020-01-18:
Can you ping your RPi  whithing your local network?

hortopan, 2020-01-18:
Yes. It connects to the network properly and I can ping it.

TOLDOTECHNIK, 2020-01-18:
What's your output of: `ssh -vvv YOUR_RASPBERRYPI_IP`

hortopan, 2020-01-18:
`
debug2: ssh_connect_direct
debug1: Connecting to 192.168.69.164 [192.168.69.164] port 22.
debug1: connect to address 192.168.69.164 port 22: Connection refused`

TOLDOTECHNIK, 2020-01-20:
There's a permission issue with SSHD. We are going to build a new image.
For reference: Create `/var/empty` folder with `root` permissions.

TOLDOTECHNIK, 2020-01-21:
A new image file is now available: https://dev.toldotechnik.li/download/387825/