TOLDOTECHNIK/buildroot-webkit #15: Overscan settings

davidenke, 2019-11-26:
Is it possible to configure over-/underscan settings with an prebuild image?
Are there settings to be set in the `config.txt` or elsewhere?

TOLDOTECHNIK, 2019-11-26:
This should work out of the box. Please refer to: https://www.raspberrypi.org/documentation/configuration/config-txt/

davidenke, 2019-11-26:
@TOLDOTECHNIK you're right it works as expected and is considered in the boot screen. But the webkit window ignores this. Is this ignored by the x server or does it require other settings?

davidenke, 2019-11-26:
The option `overscan_scale` did the trick, but as [mentioned in the documentation](https://www.raspberrypi.org/documentation/configuration/config-txt/video.md) it is not recommended as it requires the gpu to align non framebuffer alignments...

TOLDOTECHNIK, 2019-11-28:
Ok, thanks for finding this out. Unfortunately we do not have any good solution for this.

closed 2019-12-17 by TOLDOTECHNIK