TOLDOTECHNIK/buildroot-webkit #16: RPI 4 Support

err4o4, 2019-12-07:


TOLDOTECHNIK, 2019-12-07:
Compared to the new Raspberry Pi 4, the currently implemented buildroot version is not able to run on this hardware. Some patches need to be applied to make the system executable. Unfortunately we don't have enough resources to adapt the code at the moment. Hopefully there will be someone who can make the necessary adjustments.

TOLDOTECHNIK, 2020-07-22:
We are still looking for help on this one...

brunobg, 2020-10-13:
@TOLDOTECHNIK What exactly would be needed?

TOLDOTECHNIK, 2020-10-13:
The rpi4_wpe_defconfig file is needed for building the RPi4 image. We kind of started to create it: [toldotechnik_rpi4_wpe_defconfig.zip](https://github.com/TOLDOTECHNIK/buildroot-webkit/files/5373686/toldotechnik_rpi4_wpe_defconfig.zip)

brunobg, 2020-10-13:
Is this essentially the .config file generated by buildroot? What would be different compared to the default pi4 file from it?

TOLDOTECHNIK, 2020-10-21:
This fork is base on an older buildroot release. But for the RPi4 we need a newer Kernel (5.4 and above). To correctly adapt all the buildroot make/config files is a bunch of work. Maybe there's another easier solution for this.

L-vd-M, 2023-08-15:
Good day.

Is this project still active.  
Will you still be finishing the rpi4 version?

Kind regards.

TOLDOTECHNIK, 2023-08-31:
A few days ago, we checked whether an implementation is currently possible without major effort. Unfortunately, the situation has not changed.