TOLDOTECHNIK/buildroot-webkit #18: Read Only FileSystem

ghanjiboy, 2020-03-13:
Hi, I was testing this out on a Raspberry pi zero W and wanted to set things like the timezone and disable nodejs, but the filesystem says its read only.  Is there a way to temporarily remove the read only so I can make changes and reset it?

Thanks!

TOLDOTECHNIK, 2020-03-13:
Did you compile buildroot by yourself? Because on the downloadable image the root fs is read/write by default. Can you post the content of your `/etc/fstab`?

ghanjiboy, 2020-03-13:
I used the image fro pizero that was available on the Readme.

Here is the data

root@rpi0:~ # cat /etc/fstab
# <file system> <mount pt>      <type>  <options>       <dump>  <pass>
/dev/root       /               ext2    rw,noauto       0       1
proc            /proc           proc    defaults        0       0
devpts          /dev/pts        devpts  defaults,gid=5,mode=620 0      0
tmpfs           /dev/shm        tmpfs   mode=0777       0       0
tmpfs           /tmp            tmpfs   mode=1777       0       0
tmpfs           /run            tmpfs   mode=0755,nosuid,nodev  0      0
sysfs           /sys            sysfs   defaults        0       0
/dev/mmcblk0p1 /boot vfat defaults 0 0

TOLDOTECHNIK, 2020-03-13:
Ok, seems like our original file. What happens if you do a: `touch /root/somefile.txt`?

ghanjiboy, 2020-03-13:
Unfortunately it says read only file system. Is there something I could have done wrong, it's a micrsd card, no write protection and I used etcher to write the image.

TOLDOTECHNIK, 2020-03-13:
This problem is new to us. Could you please write the SD card again as described in the documentation and try again? Unfortunately we cannot give more support.

ghanjiboy, 2020-03-13:
Thanks for the assistance - I did do it twice before I reached out.  I tried building on another machine and incredibly it worked.    I was able to login and make the necessary changes to free up more memory for my use case.  Thank you for creating this - its awesome!

ghanjiboy, 2020-03-13:
Spoke too soon - it seems first boot is Read write, but subsequent boots are read only.  Is that by design?

TOLDOTECHNIK, 2020-03-14:
No, this shouldn't be this way. We tested it again and we run into the same problem.

TOLDOTECHNIK, 2020-03-14:
There must have been something going wrong with previous buildroot build. We generated a new one. Please download and try again.

ghanjiboy, 2020-03-15:
This is now confirmed working - thank you for the prompt responses and thank you for fixing and rebuilding an image.

closed 2020-03-15 by ghanjiboy