TOLDOTECHNIK/buildroot-webkit #19: Portrait mode

timemaster5, 2020-03-13:
Hi,
first thanks for a nice job.

I wanted to ask if is possible to rotate the screen, i.e. set up an accelerated portrait mode.

display_rotate in config.txt is not what I am looking for, this will result in very poor performance of the whole stack. I would like to rotate the WPE browser instead.

Any suggestions?

TOLDOTECHNIK, 2020-03-13:
This is related to https://github.com/TOLDOTECHNIK/buildroot-webkit/issues/7
What you can do is rotate the whole content of the page. Something like this: https://jsfiddle.net/egn80xh9/

timemaster5, 2020-03-13:
Yeah, slightly related.

The correct way is to set the orientation for the accelerated layer. In the case of WPEWebkit, it should be somewhere in the configuration of WPEBackend-RDK or something similar.

TOLDOTECHNIK, 2020-07-22:
closing due to no activity.

closed 2020-07-22 by TOLDOTECHNIK