TOLDOTECHNIK/buildroot-webkit #21: RPI 2 Support

itsmeow, 2020-08-10:
Some configurations for the RPi 2 would be great. I have an old one sitting around and it would be nice to put it to use. I have no idea what I'm doing, but I think I made one in my fork. I'm building it now but I have no idea if it will actually work.
For reference (feel free to use, although I'm sure you guys would know far better than me): https://github.com/itsmeow/buildroot-webkit/commit/445f4a2de9de6b50f67dc44b94b1b4a82a53e557

itsmeow, 2020-08-10:
Missed a few minor variables in my initial commit- see full diff for what I think the correct changes are https://github.com/TOLDOTECHNIK/buildroot-webkit/compare/master...itsmeow:master

itsmeow, 2020-08-11:
It built with my final changes and seems to work great! Take a look at my changes and feel free to re-create them. Don't have time to make a proper PR as I also have to remove the logo I encoded before making it, but if you're ever looking for a working RPi2 version, check my fork!

itsmeow, 2020-08-14:
Awesome! Some prebuilt images + README update are all that's left for that if you have time. Thanks for the merge.

TOLDOTECHNIK, 2020-08-14:
Thank you for sharing your changes. Unfortunately we cannot create a working image. Did you do some settings which are not available in your GitHub repo yet?

itsmeow, 2020-08-14:
As far as I know, no. That's confusing. Does it not build?

TOLDOTECHNIK, 2020-08-14:
It builds ok, but it does not boot. Could you please send/show us the content of the boot partition?

itsmeow, 2020-08-14:
Here's the boot partition of the img file.
[boot.zip](https://github.com/TOLDOTECHNIK/buildroot-webkit/files/5077394/boot.zip)

itsmeow, 2020-08-14:
Here's some system info, not sure if useful or not. As you can see it is running though...

[image]

I did make [these changes](https://github.com/itsmeow/buildroot-webkit/commit/5c8756b4c041a5951135455775bc522c44b5be3f) in the middle of my make before it had likely gotten to that point (it failed a few times due to missing libraries before actually building the OS I believe). It appears they took effect though since the system name was rpi2 in the image.

itsmeow, 2020-08-14:
I can also verify this is the most up-to-date version on my repo.
[image]

itsmeow, 2020-08-14:
I also ran `git diff HEAD > buildroot-diff.txt` on my local `buildroot` directory to get the final changes applied to the buildroot repo. There is one change I made manually while building because I had accidentally blocked inbound FTP and not realized until after I changed it, which is the change to `package/gnutls/gnutls.mk`
[buildroot-diff.txt](https://github.com/TOLDOTECHNIK/buildroot-webkit/files/5077493/buildroot-diff.txt)

TOLDOTECHNIK, 2020-08-15:
Could you please try our new build? Thanks a lot.
https://dev.toldotechnik.li/download/387854/

TOLDOTECHNIK, 2021-02-25:
Closing due to no activity.

closed 2021-02-25 by TOLDOTECHNIK

itsmeow, 2021-03-03:
Sorry, can confirm. Just tried it out, boots just fine. Acts exactly the same with my build (aside from my custom splash) vs the one linked. Thanks guys :)
[image]