TOLDOTECHNIK/buildroot-webkit #22: Is it possible to add node libs in the buildroot?

danielmorena, 2020-11-09:
I'm interested in add node GPIO libs, and omxplayer via dbus control. Is it possible extending this script? how i include this, is there a documetation?

danielmorena, 2020-11-09:
Also, is it possible to include a pre-configured terminal command? I want to switch between read-only and write-only fs modes using a bash script

TOLDOTECHNIK, 2020-11-10:
Please refer to https://github.com/TOLDOTECHNIK/buildroot-webkit/issues/9#issuecomment-486373284 for adding node.js modules.

omxplayer is already included in the prebuilt images.

TOLDOTECHNIK, 2020-11-10:
You can make the `/dev/root` read-only by editing `/etc/fstab`.
There's also an option in `make menuconfig` (`System configuration` -> `remount root filesystem read-write during boot`)

danielmorena, 2020-11-12:
excellent, i will try the receipt for the build root. Let me ask a final question, is it possible to create the image with a custom software? I mean to create an image with my app pre packed? where do i place the files?

TOLDOTECHNIK, 2020-11-12:
Best way to store custom binary files is in: `./board/toldotechnik_rpi/rootfs-overlay/usr/local/bin/`

TOLDOTECHNIK, 2020-11-21:
Have you already made progress?

danielmorena, 2020-11-23:
yes! thanks for the support!

closed 2020-11-23 by TOLDOTECHNIK