TOLDOTECHNIK/buildroot-webkit #24: not connecting to the wifi network

sahil-siwatch, 2021-02-03:
Hello. I am using RPi3B+ image as given by you guys. I have filled my wifi credentials in the wpa_supplicant.conf file but it doesn't seem to be connecting. I also tried adding the country=IN and keeping the network open but it doesn't work. Here is screenshot of my wpa file.
[image]

TOLDOTECHNIK, 2021-02-03:
Try to remove the quotes (ssid=POCO)
Ensure that the `wpa_supplicant.conf` file gets deleted on the `/boot` partition after first boot.

sahil-siwatch, 2021-02-03:
> Try to remove the quotes (ssid=POCO)
> Ensure that the `wpa_supplicant.conf` file gets deleted on the `/boot` partition after first boot.
Do I do this step before the first boot ?
I am using the prebuild images provided by you.

TOLDOTECHNIK, 2021-02-03:
1. Save the `wpa_supplicant.conf` to the boot partition.
2. Boot up your RPi
3. Check, if the config file was deleted from `/boot`

sahil-siwatch, 2021-02-03:
> 1. Save the `wpa_supplicant.conf` to the boot partition.
> 2. Boot up your RPi
> 3. Check, if the config file was deleted from `/boot`

No. It wasn't deleted. It was at the place where it was in the beginning. I edited the file in my laptop using sd card reader. I hope i am doing it correctly

TOLDOTECHNIK, 2021-02-03:
Ok, then maybe you have a wrong file name. Check if your text editor automatically adds a `.txt` filename extension.

sahil-siwatch, 2021-02-03:
> Ok, then maybe you have a wrong file name. Check if your text editor automatically adds a `.txt` filename extension.

[image]
I don't think that it is adding .txt,

TOLDOTECHNIK, 2021-02-03:
Can you please post the file's content?

sahil-siwatch, 2021-02-03:
> Can you please post the file's content?

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=IN

network={
  ssid=POCO
  psk=""
}

TOLDOTECHNIK, 2021-02-03:
You still have quotes in you config (`psk=""`). Try `psk=`

sahil-siwatch, 2021-02-03:
okay. I tried removing them as well. It doesn't work :(

TOLDOTECHNIK, 2021-02-03:
Sad. Try flashing the hole system again and take the sample config from `/boot`.

sahil-siwatch, 2021-02-03:
the procedure given here will work for rpi3b+. right ??

TOLDOTECHNIK, 2021-02-03:
Yes.

sahil-siwatch, 2021-02-03:
okay i will build it on my ubuntuwith multiple cores. fingers crossed.

sahil-siwatch, 2021-02-08:
> okay i will build it on my ubuntuwith multiple cores. fingers crossed.

Hello. I am able to run the image now. Thanks a lot for the help. Can I change the browser now i.e from WPE to any other browser ? I am using the pre-built image by you guys. Can I install new packages and other apps ??

sahil-siwatch, 2021-02-08:
> Yes.

Hello. I am able to run the image now. Thanks a lot for the help. Can I change the browser now i.e from WPE to any other browser ? I am using the pre-built image by you guys. Can I install new packages and other apps ??

TOLDOTECHNIK, 2021-02-08:
There is no other browser supported. Before compile you can run `make menuconfig` to choose other applications.

sahil-siwatch, 2021-02-09:
> There is no other browser supported. Before compile you can run `make menuconfig` to choose other applications.

So there is no change I can make if I am using your prebuilt image ? I tried doing SSH but I think busybox is not installed so I can't see anything except the root thing.

TOLDOTECHNIK, 2021-02-11:
conversation gets off topic...

closed 2021-02-11 by TOLDOTECHNIK