TOLDOTECHNIK/buildroot-webkit #26: error while building

sahil-siwatch, 2021-02-04:
Hello. I encountered some error while building the image for my rpi3b+. I followed the instructions as described in the documentation. Here is the screenshot of the error message. Can you suggest what might be causing this error ?? 
[image]

TOLDOTECHNIK, 2021-02-11:
Building is supported for Ubuntu 16.04 only for now. Please see https://github.com/TOLDOTECHNIK/buildroot-webkit#prerequisites-ubuntu-1604

sahil-siwatch, 2021-02-11:
I got the build success using Ubuntu 20.04 and 18.04 both. Just updated the e2fsprogs version and it worked.

TOLDOTECHNIK, 2021-02-11:
Ok, very good. Thanks.

closed 2021-02-11 by TOLDOTECHNIK