TOLDOTECHNIK/buildroot-webkit #27: How to enable touch display

sahil-siwatch, 2021-02-11:
hello. I have 5 inch touch screen display for rpi 3b+. https://robu.in/product/5-inch-touch-screen-hdmi-interface-display-module-tft-lcd-800x480-raspberry-pi-2-model-b-touch-pen/?gclid=Cj0KCQiAyJOBBhDCARIsAJG2h5eXf4K4tWBoQ1Q2Y2B9ikgrrpC_3CLkp7SBnB5iyif-6HcWIMHjPgkaAr0AEALw_wcB

How do I enable the touch input ?

TOLDOTECHNIK, 2021-02-11:
You need to setup a special driver for the touch screen located here: https://github.com/goodtft/LCD-show
Unfortunately we cannot give you more support.

sahil-siwatch, 2021-02-11:
Okay. Can't you guide me how can I setup third party softwares in your image that I built ??

TOLDOTECHNIK, 2021-02-25:
Sorry, we cannot provide any further support.

closed 2021-03-05 by TOLDOTECHNIK