TOLDOTECHNIK/buildroot-webkit #28: Rpi zero gpio access

ghanjiboy, 2021-02-25:
Hi, first wanted to say thank you, the rpi zero image works great.

I am looking to add a motion sensor to the setup using gpio and not sure what the right way to do that as there isn't any python and the node version doesn't support onoff.  Do you have any recommendations?

TOLDOTECHNIK, 2021-02-25:
Did you see https://github.com/TOLDOTECHNIK/buildroot-webkit/issues/9?

ghanjiboy, 2021-02-26:
I did, but wasnt sure its same response as I am asking about the zero.  I was actually able to do this C++ by cross compiling and using wiring PI.   So now I can get to the gpio pins.  The last part now is about turning off the display - i was looking to use tvservice which just doesnt seem to work - it just hangs.   I tried copying a version from rpi3 and also cross compiling.  i can get the tvservice to show help info, but not when passing any parameters - it just sits there.   Do you have any idea about this or an alternate way to turn the display off/on?

Thanks!

ghanjiboy, 2021-03-20:
I was able to solve this.   The use case is for a kiosk where if there wasnt any motion, it would disable the screen and enable if there was.  I used WiringPi library in a C program with a motion sensor and used the already existing command vcgencmd with parameter display_power to control the screen.

closed 2021-03-20 by ghanjiboy

mindnuts, 2021-04-14:
@ghanjiboy care to share your work here? i am also interested in such a solution. Its for a waiting room. If there is no movement i would like the Screen to go off after lets say 10 minutes.

ghanjiboy, 2021-04-15:
[RPiSleepMotion.zip](https://github.com/TOLDOTECHNIK/buildroot-webkit/files/6314563/RPiSleepMotion.zip)

Find attached some work - here's my setup:

Requirements: WSL and Docker if on windows - or you can run this on a linux box too, just might need some modifications

1. I used a cross compiler to build the C program - I used Ubuntu on WSL for Windows - thats under the cross-compiler folder - run the build.sh command which will download the docker image from docker hub for dockcross/linux-armv6
2. #1 should be  a one time activity
3. Then its time to look at the C program itself - its super rudimentary and just is orchestrating vcgencmd - alter as you see fit.  Just run the build.sh command which downloads WiringPi, compiles it as a dependency