TOLDOTECHNIK/buildroot-webkit #30: Change resolution

tonyjobson, 2021-04-14:
Hi, 

I added 
"hdmi_mode=2"
to the end of /boot/config.txt but it has not adjusted the default resolution of the system.

Can you offer any other pointers on how I can configure resolution?

My screen accepts a 1080p signal and samples it down but it leads to very small text on a 4 inch screen. I'd like to lock the pi at native resolution of the my output device so text is readable.

thanks

TOLDOTECHNIK, 2021-04-14:
Did you also define the `hdmi_group`?

tonyjobson, 2021-04-14:
I had not. thank you for the pointer.

I first tried hdmi_safe=1 and this worked to some extent.

Then I went to 
hdmi_group=2
hdmi_mode=14

This has set my screen (480) properly and things are now working great!

Thank you for your help!

closed 2021-04-14 by tonyjobson