TOLDOTECHNIK/buildroot-webkit #31: Page not found solution

doublec4, 2021-04-29:
Hello, in the event that the network drops and the page is not found... is there some kind of custom "page not found" that can be used? Or perhaps switch over to the local webserver to display a custom page?

I am new to Raspberry Pi and I would love to use this amazing creation of yours to show my PLC eHMI touchscreen interface for a home automation system I am building. For instance, I would like to have a touchscreen on each floor of the house to control things like lights etc. The master PLC with the eHMI webserver would be in a cabinet in the basement. Therefore the touchscreens would point to the IP address of this webpage from the master PLC to boot to on start up.

However, in the event that my local network goes down, I would like to have an error screen / page, something that provides the user with some notice such as "network issue detected. Press button to override and turn off all local devices."

Therefore, the onscreen button that could be pressed could perhaps switch one of the GPIO pins on the raspberry pi which would be directly connected to the local DIO relays which are in the same physical location. This would give the local relay board the command to then kill all active local relays and shut everything down locally. Therefore my relays could at least still be deactivated and not rely on the network connection through the network interface.

Do you think something like this is possible to achieve? 

Thank you

TOLDOTECHNIK, 2021-04-29:
This is a little bit tricky to realize. There are two ways to achieve what you want.

1. Run a task that checks the internet connection every minute or so. If connectivity fails, kill the `wpe` browser process and run it again with a custom URL: `/usr/bin/wpe --loop http://localhost/404.html`
2. Create a local hosted web page that nests your external web page in an iframe. Within this local hosted web page you can run a script that checks connectivity and reload the iframe's content with another page when needed.

TOLDOTECHNIK, 2021-05-15:
Have you made any progress in this case?

doublec4, 2021-05-16:
Not yet, been focusing on my PLC program lately but will try to address
this soon.

TOLDOTECHNIK, 2021-11-03:
Closing due to no updates.

closed 2021-11-03 by TOLDOTECHNIK