TOLDOTECHNIK/buildroot-webkit #34: Web inspector encountered an internal error

TheLaGmAn, 2021-07-24:
Im testing this build (actually the release image) in a RPI3.
When using the legacy inspector (http) i always get this errors 24/7 on almost every action on it, making it unusable.
Is this not functional with toldotechnik?
Also, there is no way to enable the not-legacy web inspector? it opens the port but midori cant connect to it, doesnt show anything to inspect

TOLDOTECHNIK, 2021-07-26:
Hi, we do not understand your question. A so called "inspector" is not supported.

TheLaGmAn, 2021-07-30:
i meant this part as described in the readme
-------------
You can call the web inspector from an external client browser (tested with Chrome and Safari, not working with Firefox). http://rpi0.local:9998 or http://rpi3.local:9998

If you want do disable it, you can comment out the export WEBKIT_LEGACY_INSPECTOR_SERVER=0.0.0.0:9998 line in /usr/bin/wpe
-------------
Accessing http://x.x.x.x:9998 throws errors 24/7 "web inspector encountered an error", making the inspector unusable, and if i enable 
WEBKIT_INSPECTOR_SERVER=0.0.0.0:9998 (the one that should be used with midrori as inspector://x.x.x.x:9998 it does not find anything to inspect

TOLDOTECHNIK, 2021-07-30:
You need to use Chrome or Safari.

TheLaGmAn, 2021-07-30:
Yea, i did, on chrome it does not show anything inside the tabs, thats why i moved into firefox which worked better considering how it works on chrome, maybe it doesnt work anymore with the current chrome version.
Do you know if there is a way to enable the not-legacy version? WEBKIT_INSPECTOR_SERVER

TOLDOTECHNIK, 2021-07-31:
It's just working on our machines:
[image]

Can you please provide more information?

TOLDOTECHNIK, 2021-11-03:
Did you have success on this?

TheLaGmAn, 2021-11-05:
only got it to work on edge, FF and chrome does not work and the non legacy method does not work either

TOLDOTECHNIK, 2021-11-05:
Ok, good news, thanks. Closing for now.

closed 2021-11-05 by TOLDOTECHNIK