TOLDOTECHNIK/buildroot-webkit #35: Automatic shutdown possible?

momochihammer, 2021-09-20:
Is there a way to have the buildroot-webkit automatically shutdown daily?

Thanks in advance.

TOLDOTECHNIK, 2021-09-21:
You can create a crontab job that runs every day at a specifig time.

Unfortunately we did not implement it in the current setup by default, so you need to do a little bit of work.

1. Create an init.d script to run the cron service on startup:

`nano /etc/init.d/S90crond`

```
#!/bin/sh

CRON_JOB="30 22 * * * /root/poweroff.sh"

case "$1" in
        start)
                printf "Starting cron ... "
                mkdir -p /var/spool/cron/crontabs
                echo "$CRON_JOB" > /var/spool/cron/crontabs/root
                start-stop-daemon -S -q -m -b -p /var/run/dcron.pid --exec /usr/sbin/crond -- -f
                echo "done."
                ;;
        stop)
                printf "Stopping cron ..."
                start-stop-daemon -K -q -p /var/run/dcron.pid
                echo "done."
                ;;
        restart)
                $0 stop
                sleep 1
                $0 start
                ;;
        *)
                echo "usage: $0 {start|stop|restart}"
                ;;
esac
```

To make it simple, the cron job definition is also stored in the init script. For crontab scheduler definitions you can look at [www.crontabgenerator.com](http://www.crontabgenerator.com/) for easily create the desired pattern.

2. Make the init.d script executable:
`chmod +x  /etc/init.d/S90crond`

3. Create the shutdown script:

`nano /root/poweroff.sh`

```
#!/bin/sh
poweroff &
exit 1
```

BTW: Putting the script into the root folder may not the best choice, but we leave it for now.

4. Make the shutdown script executable:
`chmod +x /root/poweroff.sh`

5. Reboot your system.

Ensure to have the system time set correctly (e.g. internet sync) - else you have to setup the cron job with a delta t.

momochihammer, 2021-09-25:
Thanks for providing such fine details. From the information I gleaned from the web, it seems that I shouldn't really need to shutdown the RPi so I didn't use your script. I'm sure it would have done a great job. 

Thanks again.

closed 2021-09-25 by momochihammer