TOLDOTECHNIK/buildroot-webkit #36: Is there ability to use HyperPixel

HaywoodSolutions, 2021-09-26:
I am trying to get this working on a HyperPixel display. It requires running, "curl -sSL https://get.pimoroni.com/hyperpixel4 | bash" but these seem unavailable? Anyone know if I could do this?

TOLDOTECHNIK, 2021-09-26:
We assume that you are using a RPi 3. If so, you can try the following.

On your running system:
- Copy/save the `/boot/overlays/hyperpixel4.dtbo` from your running system

On the buildroot-webkit environment:
- To access the `boot` partition mount it with `mount /dev/mmcblk0p1 /mnt`
- Copy the previously saved `hyperpixel4.dtbo` into the `/mount/overlays` folder
- Add the lines below into the `/mount/config.txt` file as defined here: https://github.com/pimoroni/hyperpixel4/blob/b71d7805e796cc67449164887cfd01f8607a6bc0/install.sh#L15-L26
```
dtoverlay=hyperpixel4
overscan_left=0
overscan_right=0
overscan_top=0
overscan_bottom=0
enable_dpi_lcd=1
display_default_lcd=1
display_rotate=0
dpi_group=2
dpi_mode=87
dpi_output_format=0x7f216
hdmi_timings=480 0 10 16 59 800 0 15 113 15 0 0 0 60 0 32000000 6
```

After that reboot the system. We don't know if the steps above work because the `dtbo` got compiled on a different system...

TOLDOTECHNIK, 2021-11-03:
Did you have success on this?

TOLDOTECHNIK, 2021-12-09:
Closing due to no activity.

closed 2021-12-09 by TOLDOTECHNIK