TOLDOTECHNIK/buildroot-webkit #37: Greek language support

ligantx, 2021-11-02:
I try to load a website that is in Greek. But, instead of the greek letters, i see squares. Can it be fixed?

TOLDOTECHNIK, 2021-11-03:
Can you please provide the URL you want to load?

TOLDOTECHNIK, 2021-11-03:
You can send it to our company's info email address. Does your html has some encoding meta declaration in the head section (`<meta charset="utf-8"/>`)?

ligantx, 2021-11-08:
email sent
and no...there is no `utf-8` declaration

TOLDOTECHNIK, 2021-11-10:
You have the declaration already in the html document: `<META http-equiv="Content-Type" content="text/html; charset=utf-8">` But we did not get the page displayed correctly. This probably have something to do with the webkit browser. We cannot fix it for now.

ligantx, 2021-11-10:
I see..thank you very much :)

closed 2021-11-11 by TOLDOTECHNIK