TOLDOTECHNIK/buildroot-webkit #38: Change the boor screen

tomkruk, 2021-11-05:
Is there some way to change or make completely blank the boot screen?

TOLDOTECHNIK, 2021-11-05:
We did not test it, but you can try to remove the `/usr/bin/psplash` program.

tomkruk, 2021-11-05:
This DOES work. Thank You. However I can see now that it can be an issue since the black screen can be misunderstood for things not working or being connected. 

Is there a way to change the image into another? Or does the whole thing need to be reassembled?

I am running a small media company where we are trying to have some digital signage. I would ideally want to put a logo and my contact information (in the image), that would be ideal. 

I do not mean to give you guys work but is there a way you can help me out? I am happy to pay for this assistance or advice.

TOLDOTECHNIK, 2021-11-05:
Please refer to  #3

closed 2021-11-11 by TOLDOTECHNIK