TOLDOTECHNIK/buildroot-webkit #41: i want NGINX/PHP image for "RPi0 w" please

Rdivxe, 2021-12-07:
i want NGINX/PHP image for "RPi0 w" please

TOLDOTECHNIK, 2021-12-09:
We'll see if we can achieve that without a lot of effort.

Rdivxe, 2021-12-09:
@TOLDOTECHNIK Wait for the img php/nginx rpi0w file. Thank you for the help. I hope to work on it as soon as possible.

TOLDOTECHNIK, 2021-12-09:
We have noticed that we have already provided a detailed setup guide. Please do the following:
Go through all the steps described [here](https://github.com/TOLDOTECHNIK/buildroot-webkit#how-to-build-it-manually). Select the configuration for the RPi0 `make toldotechnik_rpi0_wpe_defconfig`. Before compiling the image follow the steps described [here](https://github.com/TOLDOTECHNIK/buildroot-webkit/issues/13#issuecomment-529283054).

Rdivxe, 2021-12-09:
@TOLDOTECHNIK These steps are difficult, I can't do it, how do I type make, where do I do it and what kind of operating system is required

TOLDOTECHNIK, 2021-12-09:
You need a linux based system. Preferably Ubuntu 16.04, because we can guarantee it works.
On a 2.3 GHz Quad-Core Intel Core i5 the build process takes about 2-3 hours.

Don't know if buildroot compilation tasks work on the Raspberry OS. If it does, the build process will take forever.

Rdivxe, 2021-12-09:
@TOLDOTECHNIK I will give you $3 to make a copy of rpi0w php nginx I will be grateful to you because my capabilities do not allow you to separate me Mr Toldo please help me and make img php nginx file

TOLDOTECHNIK, 2021-12-09:
Your colleague has already contacted us as well. We have sent him our conditions for commercial use. We are closing the issue because it is no longer relevant for this project.

closed 2021-12-09 by TOLDOTECHNIK