TOLDOTECHNIK/buildroot-webkit #42: Magic Mirror

momochihammer, 2021-12-23:
Hi - Sorry for the newbie question, but can Magic Mirror for Raspberry Pi be installed on this? I like the lightweight of this since I'm running a Pi zero w.

TOLDOTECHNIK, 2021-12-23:
Magic Mirror is encapsulated in Electron. For it to run, an X11 server is required. This is not part of our system.

momochihammer, 2021-12-24:
Thanks for answering so quickly. I really appreciate it.

TOLDOTECHNIK, 2021-12-27:
We did a little research on the Magic Mirror application. Here's what you can do:

- Run the Magic Mirror in server mode on a host device (https://docs.magicmirror.builders/getting-started/installation.html#manual-installation)
- Point your Raspberry Pi to the host (https://github.com/TOLDOTECHNIK/buildroot-webkit#url-for-the-browser)

momochihammer, 2021-12-27:
Thanks for doing the extra research for this. I hope this didn't take too much of your time. Will try to see if I can use a host device for the MM.  Thanks again for all your help.

closed 2021-12-27 by momochihammer