TOLDOTECHNIK/buildroot-webkit #44: Custom boot image

Ulisseruiz, 2022-05-10:
Hi, can you write step to change boot image custom or blacked?

TOLDOTECHNIK, 2022-05-11:
Please refer to: https://github.com/TOLDOTECHNIK/buildroot-webkit/issues/3
For a black screen you can just delete the `psplash` executable.

Ulisseruiz, 2022-05-11:
I view you can make this with money and no problem. But i testing all and with 4 break power SD are corrupted. You insert precaution for this inside? or setting only read image? Thank you

TOLDOTECHNIK, 2022-05-11:
Sorry, but we do not understand your question.

Ulisseruiz, 2022-05-11:
i try your image for sd card but when have discontinued power supply sd card corrupted system files. 
Inside in your sd card image your prevent this?

TOLDOTECHNIK, 2022-05-11:
This never happened on our installations. Please try to use another SD card.

TOLDOTECHNIK, 2022-05-16:
Did you solve the problem?

denizyesilirmak, 2022-05-16:
I think you should use a better sd card. I don't think it will corrupt the filesystem unless you turn off the power during a heavy file write operation. It never occurred to me.

TOLDOTECHNIK, 2022-06-04:
Closing due to lack of activity.

closed 2022-06-04 by TOLDOTECHNIK