TOLDOTECHNIK/buildroot-webkit #47: i get this error when i am compiling

pablopolanco6, 2022-08-04:
CC       libgio_2_0_la-thumbnail-verify.lo
gdbusauth.c: In function ‘_g_dbus_auth_run_server’:
gdbusauth.c:1304:11: error: ‘%s’ directive argument is null [-Werror=format-overflow=]
 1304 |           debug_print ("SERVER: WaitingForBegin, read '%s'", line);
      |           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [Makefile:3547: libgio_2_0_la-gdbusauth.lo] Error 1
make[5]: *** Waiting for unfinished jobs....
gdbusmessage.c: In function ‘parse_value_from_blob’:
gdbusmessage.c:1706:29: warning: variable ‘item’ set but not used [-Wunused-but-set-variable]
 1706 |                   GVariant *item;
      |                             ^~~~
gdbusmessage.c: In function ‘append_value_to_blob’:
gdbusmessage.c:2322:24: warning: unused variable ‘end’ [-Wunused-variable]
 2322 |           const gchar *end;
      |                        ^~~
gdbusmessage.c: In function ‘g_dbus_message_to_blob’:
gdbusmessage.c:2698:30: error: ‘%s’ directive argument is null [-Werror=format-overflow=]
 2698 |       tupled_signature_str = g_strdup_printf ("(%s)", signature_str);
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [Makefile:3596: libgio_2_0_la-gdbusmessage.lo] Error 1
make[5]: Leaving directory '/home/azureuser/buildroot/output/build/host-libglib2-2.50.3/gio'
make[4]: *** [Makefile:4370: all-recursive] Error 1
make[4]: Leaving directory '/home/azureuser/buildroot/output/build/host-libglib2-2.50.3/gio'
make[3]: *** [Makefile:1994: all] Error 2
make[3]: Leaving directory '/home/azureuser/buildroot/output/build/host-libglib2-2.50.3/gio'
make[2]: *** [Makefile:1213: all-recursive] Error 1
make[2]: Leaving directory '/home/azureuser/buildroot/output/build/host-libglib2-2.50.3'
make[1]: *** [Makefile:855: all] Error 2
make[1]: Leaving directory '/home/azureuser/buildroot/output/build/host-libglib2-2.50.3'
make: *** [package/pkg-generic.mk:219: /home/azureuser/buildroot/output/build/host-libglib2-2.50.3/.stamp_built] Error 2

TOLDOTECHNIK, 2022-08-04:
We can't give you support for that. Please refer to: [Build manually](https://github.com/TOLDOTECHNIK/buildroot-webkit#how-to-build-it-manually)

TOLDOTECHNIK, 2022-08-22:
Closing due to no activity.

closed 2022-08-22 by TOLDOTECHNIK