TOLDOTECHNIK/buildroot-webkit #52: Wifi connection required for webserver

jbrangwyn, 2022-10-06:
Hi, I had a raspberry pi 2 running a locally served HTML page on the webserver. It does not have built in WiFi, so I was using a usb WiFi dongle.

When I remove the dongle, I only get a white splash screen. This was going to be used somewhere without a WiFi connection.

Is WiFi/network connection required to run the web server? Thanks.

TOLDOTECHNIK, 2022-10-06:
What application is serving your HTML page? Is it running on the board using our Node.js Express server (`/var/node/server.js`)?

jbrangwyn, 2022-10-06:
Yes, using the node.js express server.

TOLDOTECHNIK, 2022-10-06:
The virtual local loopback interface (lo) should handle all the localhost requests without any other network interfaces activated at all. So the web page should be displayed.
Ok, we will try our buildroot-webkit on a RPi 2 without any network connections the next days...

TOLDOTECHNIK, 2022-10-08:
We do not have access to a RPi 2 right now. But we tested your issue on a RPi B+
The browser is displaying the default locally hosted web page with no problems (no ethernet, no WiFi dongle attached).
Can you please try our [image](https://dev.toldotechnik.li/download/387854/) without any modification again? Can you see the rotating squares then?

TOLDOTECHNIK, 2022-10-08:
Ok, found it. The browser gets started too early when no network is active. That means, the browser is running before the local webserver (Node.js) is started.

Try to add a slight delay in the `/etc/init.d/S90wpe` script. Something like this:
[image]

closed 2022-10-10 by jbrangwyn

denizyesilirmak, 2022-10-10:
If you don't want to wait 5 secods, you can watch your server's port and then launch WPE.

start)
    /usr/bin/node /var/node/index.js > /dev/null 2>&1 &
    # Wait for server to open WPE
    while ! `netstat -tln 2>/dev/null | grep -q  8080`;
    do
      sleep 0.3;
    done
    echo -n "Starting WPE: "
    /usr/bin/wpe --loop "http://localhost:8080" &> /dev/null &
    ;;
  stop)