TOLDOTECHNIK/buildroot-webkit #53: Read-only filesystem for RPi3?

Lee181, 2023-04-28:
Hello guys,

thank you very much for this great project!!

I have used your prebuilt image for RPi3 (https://dev.toldotechnik.li/download/sdcardRPi3.img.zip) from 2019-06-24 on an RPi3 A+, everything seems to work fine.

But it seems, the filesystem is NOT read-only, can you check that please?

I stumbeled over https://github.com/TOLDOTECHNIK/buildroot-webkit/issues/18 
There you provided a new prebuilt image for Pi Zero one day before the issue got closed.

Is it possible (if I look at the dates of the prebuilt images), that you forgot to apply the same fix like in #18 for RPi3?

Here is my fstab output:
`root@rpi3:~ # cat /etc/fstab
# <file system>	<mount pt>	<type>	<options>	<dump>	<pass>
/dev/root	/		ext2	rw,noauto	0	1
proc		/proc		proc	defaults	0	0
devpts		/dev/pts	devpts	defaults,gid=5,mode=620	0	0
tmpfs		/dev/shm	tmpfs	mode=0777	0	0
tmpfs		/tmp		tmpfs	mode=1777	0	0
tmpfs		/run		tmpfs	mode=0755,nosuid,nodev	0	0
sysfs		/sys		sysfs	defaults	0	0
/dev/mmcblk0p1 /boot vfat defaults 0 0
`
Can you have a look at this, please?

By the way: Is it possible to provide an additional prebuilt image configured for RPi 3A+ (only 512MB RAM, no LAN)?

For the moment, the RPI3 B+ image seems to work well, but I don't know, what happens if the read-only mode gets activated due to the half amount of RAM?

I have another wish/idea/question:
Would it be possible to set an individual hostname (for network AND mDNS/Avahi) somehow via config.txt?
I would need three RPis in the same network with different content, so I can identify/access them via mDNS.

Thanks in advance,
 Micha

TOLDOTECHNIK, 2023-05-02:
By default we only provide R/W images. You can modify `/etc/fstab` so the `/dev/root` gets mounted read only.

For to have some other network names you simply have to change it in the `/etc/hostname` file.

Lee181, 2023-05-02:
Thank you for your reply!

I have already changed
`/etc/fstab` 
to 
`/dev/root	/		ext2	ro,noauto	0	1`
before I opened this issue, but i still can write/edit around on /dev/root after reboot.

Even If I try `mount -o remount,rw /`, I can still write.

What am I doing wrong?
I am just looking for a way to be completely read-only for less wear of the SD-Card...

TOLDOTECHNIK, 2023-05-03:
Ok, we had a deeper look into this. It is not possible to have a read only filesystem by just changing the `/etc/fstab` configuration file. It needs a "file system overlay" so that the linux system works correctly. But for this an adapted version of the RPi image is necessary. We do not provide such.

Lee181, 2023-05-08:
Thank you very much for digging in and for clarification, guys!
Herzliche Grüße nach Vaduz!