TOLDOTECHNIK/buildroot-webkit #54: Rotate display

invisiblestupid, 2023-08-15:
This app is a dream come true for me.  All the others I've found bog down on the Pi Zero.  It works great with Google Slides as my presentation tool.  However, Is there a way to rotate the display so that it will run in portrait mode?  I can rotate the content on the slides...it's just a little harder to maintain the content.  I'm sure I look funny tilting my head like that to see the content.  :)

Also, in one application, I'm not going to have wifi available.  I know I can create a video of the presentation but that's a big cumbersome too.  If you're still working on this project at all, I'd love to see an image.txt or some such to point to a folder of images to loop through for a pre-determined length per image.

Even if you don't get to it, this is an amazing little tool and will be using it in my store.

invisiblestupid, 2023-08-16:
Thanks!  The rotation trick worked.

I think I'll just stick with a video and be done with it.  Thanks!

TOLDOTECHNIK, 2023-12-30:
Closing due to no activity.

closed 2023-12-30 by TOLDOTECHNIK